AstroWimSara/SolarEclipseWorkbench #79: Installer

WimDeMeester, 2025-06-15:
Installing the Solar Eclipse Workbench is not easy. We should check if we can create an installer to make this easier.

astronerd1, 2025-06-15:
Please do this! After watching your presentation on June 13th, I downloaded your program stuff to experiment with my cameras but was confronted with a difficult installation process. Even though I am a retired programmer in Mainframe IBM360 for 30 years, I will wait to do this installation. I guess I will rely on my old MacBook and SEM until this software has an installation package.

On another note, Can you add to the script commands the "LAST" command? It works like this:

TAKEPIC,C2,+,00:00:05.0,D850,1,8.0,100,0.000,RAW,,Y,
TAKEPIC,LAST,+,00:00:02.0,D850,1,8.0,100,0.000,RAW,,Y,

where the first script line is executed 5 seconds after C2 and the second script line is executed 2 seconds after the LAST script line.

WimDeMeester, 2025-06-16:
Thanks @astronerd1 , 
We will try to make this high priority!  I will also add a new issue on the LAST command.

closed 2026-01-15 by WimDeMeester